ARM-software/ML-KWS-for-MCU #96: Calculate a negative number in bias_shift (shift right)

ccnankai, 2019-03-24:
At the final fc layer bias dec bit =10. I calculated the last layer, wx=8  ，  bias=10 ， bias_shit=-2，bias need to be shifted right 。 But there is no option for the right shift 。 what should I do with this situation.
[image]

saichand07, 2019-03-25:
@ccnankai 
could you tell me in where did you find the "final fc layer bias dec bit =10. I calculated the last layer, wx=8 ， bias=10 ， bias_shit=-2"
can you put the link to the code (lines)

ccnankai, 2019-03-26:
last inputlayer x w =8bit dec      bias=10  ====> bias_shit=-2

ccnankai, 2019-03-26:
@saichand07 
If I change clip+ms to 2000, which parameters should be adjusted accordingly.
average pooling and fc layer ，Are the parameters of these two layers changed?

saichand07, 2019-03-26:
@ccnankai 
how to calculate bias_shift, out_shift (left, right) manually?

ccnankai, 2019-04-03:
@saichand07 
Sorry, I am busy recently. The calculation step has been told in this question#51.Check out the quant_guide and re-read the quantization section again.The steps are simple.Discuss problems together.swdtianxia@gmail.com

saichand07, 2019-05-07:
@ccnankai 

I have calculated all the bias_shifts and out_shifts layer by layer by adjusting the input and output format (using act_max ),  like they explained in the quant guide. I put those values in ds_cnn.h
and I also tuned the recording window, averaging window len and detection threshold. 
but still deployed model on board is not detecting words correctly.

could you please tell me what else I need to taken care of.
how did you deal with negative shift ?

Thank you very much

ccnankai, 2019-05-09:
@saichand07 
Did you calculate the MFCC_DEC_BITS ? How did you calculate it.Are you calculating the other *shift parameters correctly?

saichand07, 2019-05-09:
yes, I calculated shift parameters as they explained in the quant guide,
How to calculate MFCC_DEC_BITS ?
Actually, I didn't change its value, I put MFCC_DEC_BITS = 1 and also changed it to 2, but still its detecting wrong words

saichand07, 2019-05-09:
@ccnankai 
my first layer Q5.2
weights Q-3.10
bias Q1.6
output layer Q3.4

BIAS_RSHIFT = 2+10-6 = 6
OUT_LSHIFT = 2+10-4 = 8

I calculated all layers like this,

last layer I got negative shift  and I don't know how to deal with it, 
Pool layer act_max I took Q3.4 and last layer Q5.2 so shift of pool layer is 4-2 = 2

could you tell me whatelse I need to consider ?

ccnankai, 2019-05-09:
@saichand07 
I think you still have the wrong calculation of the parameters.I can see good results.

saichand07, 2019-05-09:
Could you please tell me how to calculate it ?

ccnankai, 2019-05-09:
Weights Q-3.10 Wrong, how did you get it?

saichand07, 2019-05-09:
after running Quant_test.py the weights are generated randomly,
my parameters are
window size_ms 40
window_stride_ms 20
dct_coefficient_count 10

saichand07, 2019-05-09:
do i need to change any thing in fold_batchnorm.py,
but results in python are like train 93.7%, validation 92.6, test accuracy 92.5%

ccnankai, 2019-05-09:
Weights Q-3.10 This parameter is definitely wrong. No need to change any thing in fold_batchnorm.Are you a student, do you use the lab for experiments?

saichand07, 2019-05-09:
yes I am student, I don't use any lab, I am writing my thesis, how to deal with this situation ?

ccnankai, 2019-05-09:
How did you get this result Weights Q-3.10

saichand07, 2019-05-09:
I trained my model and tested, I used to fuse my checkpoints using fold batchnorm.py, then I run Quant_test.py, I didn't make any changes 

sometimes weights are generated like Q-2.9, still its wrong I guess ?
is that possible for you to share your parameters ? and can you give me some ideas, so that I can get rid of thi

saichand07, 2019-05-09:
int_bits-3
dec_bits10
DS-CNN_conv_1_weights_0 number of wts/bias: (10, 4, 1, 64) dec bits: 10 max: (0.072265625,0.072254896) min: (-0.08300781,-0.08279022)
int_bits1
dec_bits6
DS-CNN_conv_1_biases_0 number of wts/bias: (64,) dec bits: 6 max: (0.9375,0.9329899) min: (-1.578125,-1.5782655)
int_bits2
dec_bits5
DS-CNN_conv_ds_1_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (3.90625,3.8910677) min: (-3.96875,-3.9709406)
int_bits2
dec_bits5
DS-CNN_conv_ds_1_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (1.96875,1.9795195) min: (-2.59375,-2.5947583)
int_bits0
dec_bits7
DS-CNN_conv_ds_1_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.6875,0.68623555) min: (-0.640625,-0.6382886)
int_bits2
dec_bits5
DS-CNN_conv_ds_1_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.03125,2.0291393) min: (-1.84375,-1.8481525)
int_bits2
dec_bits5
DS-CNN_conv_ds_2_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (1.96875,1.9547772) min: (-2.25,-2.2396395)
int_bits2
dec_bits5
DS-CNN_conv_ds_2_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.1875,2.1813045) min: (-1.34375,-1.347389)
int_bits0
dec_bits7
DS-CNN_conv_ds_2_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.578125,0.57834643) min: (-0.671875,-0.6702178)
int_bits2
dec_bits5
DS-CNN_conv_ds_2_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.84375,2.84532) min: (-1.4375,-1.4403846)
int_bits2
dec_bits5
DS-CNN_conv_ds_3_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (2.25,2.249133) min: (-1.9375,-1.9425142)
int_bits2
dec_bits5
DS-CNN_conv_ds_3_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.1875,2.1995513) min: (-1.71875,-1.7092787)
int_bits0
dec_bits7
DS-CNN_conv_ds_3_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.7109375,0.7123333) min: (-0.6171875,-0.6133568)
int_bits1
dec_bits6
DS-CNN_conv_ds_3_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.59375,1.5910984) min: (-1.625,-1.6272246)
int_bits2
dec_bits5
DS-CNN_conv_ds_4_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (2.03125,2.0189984) min: (-1.75,-1.74682)
int_bits2
dec_bits5
DS-CNN_conv_ds_4_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (1.4375,1.450733) min: (-2.4375,-2.4516432)
int_bits0
dec_bits7
DS-CNN_conv_ds_4_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.796875,0.7958698) min: (-0.75,-0.7508208)
int_bits1
dec_bits6
DS-CNN_conv_ds_4_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.9375,1.9307799) min: (-1.03125,-1.0330939)
int_bits2
dec_bits5
DS-CNN_fc1_weights_0 number of wts/bias: (64, 12) dec bits: 5 max: (1.75,1.7561988) min: (-2.84375,-2.8390794)
int_bits-1
dec_bits8
DS-CNN_fc1_biases_0 number of wts/bias: (12,) dec bits: 8 max: (0.3515625,0.35339496) min: (-0.26171875,-0.26149127)
INFO:tensorflow:set_size=36923
INFO:tensorflow:Confusion Matrix:
 [[3118    0    0    0    0    0    0    0    0    0    0    0]
 [  12 2530   16   63   34   39   40   43  118   15   84   62]
 [   6   53 3139   37    3    9   20   11    1    1    1    1]
 [   7   39    6 3045    2    9    8    5    2    0    9   55]
 [  11   46    0   12 2669    1   10    8   22  119   68   19]
 [  15   53    8  143    5 2784   10    7   23    1   47   93]
 [   6   54   52   25   15    1 2817   34    1    5    6    7]
 [   3   93    4    5    6    4   40 2844    7    2    1    2]
 [   2   50    0    8   33    4    4    4 2904   39   13   11]
 [   7   31    0    6  107    0    8    3   95 2695   19   15]
 [   5   29    0    4   35    4    7    5    1    5 2890   17]
 [   7   97    3  132    7   26    5   11    0    9   35 2680]]
INFO:tensorflow:Training accuracy = 92.39% (N=36923)
INFO:tensorflow:set_size=4445
INFO:tensorflow:Confusion Matrix:
 [[371   0   0   0   0   0   0   0   0   0   0   0]
 [  0 307   2   9   2  11   6   4   7   3   9  11]
 [  0   7 376   2   0   1   6   2   0   0   1   2]
 [  1  11   2 384   0   1   1   1   0   0   1   4]
 [  0   9   0   1 314   0   3   1   0  10  11   1]
 [  1   2   1  19   0 340   2   1   1   0   4   6]
 [  0   7  13   2   2   0 322   4   0   1   1   0]
 [  0  11   0   1   0   0   4 345   1   0   0   1]
 [  1   6   0   0   6   2   0   1 339   7   1   0]
 [  0   3   0   1  25   0   1   0   9 330   2   2]
 [  3   3   0   0   9   0   0   1   1   1 331   1]
 [  1  14   0  18   1   3   1   2   2   1   4 325]]
INFO:tensorflow:Validation accuracy = 91.88% (N=4445)
INFO:tensorflow:set_size=4890
INFO:tensorflow:Confusion Matrix:
 [[408   0   0   0   0   0   0   0   0   0   0   0]
 [  0 332   1  10   3   6   6   4  22   4   8  12]
 [  0   9 398   6   0   1   1   1   0   0   2   1]
 [  0   5   0 382   0   2   2   0   0   1   1  12]
 [  2   7   0   0 386   0   1   0   3  11  14   1]
 [  0  10   2  24   0 356   2   2   0   1   3   6]
 [  1   7  10   1   1   0 389   3   0   0   0   0]
 [  1  18   0   0   0   0   8 368   0   0   1   0]
 [  1   6   0   0   7   1   1   1 363  12   1   3]
 [  1   2   0   2  16   0   1   0  10 359   0  11]
 [  1   3   0   0   6   1   0   0   0   2 393   5]
 [  2  15   0  34   1   0   3   1   1   2   1 342]]
INFO:tensorflow:Test accuracy = 91.53% (N=4890)

ccnankai, 2019-05-09:
Sorry, my English is not good, I said how to calculate q2.9. From which value is obtained. weights don't need calculate.

saichand07, 2019-05-09:
depthwise_conv = slim.separable_convolution2d(inputs,
                                                  num_outputs=None,
                                                  stride=stride,
                                                  depth_multiplier=1,
                                                  kernel_size=kernel_size,
                                                  scope=sc+'/dw_conv',
                                                  reuse=tf.AUTO_REUSE)
    if(act_max[2*layer_no]>0):
      depthwise_conv = tf.fake_quant_with_min_max_vars(depthwise_conv, 
          min=-act_max[2*layer_no], 
          max=act_max[2*layer_no]-(act_max[2*layer_no]/128.0), 
          num_bits=8, name='quant_ds_conv'+str(layer_no))
    
      min_value = -act_max[2*layer_no]
      max_value=act_max[2*layer_no]-(act_max[2*layer_no]/128.0)
      print('quant_ds_conv_min_value' +str(min_value), 'quant_ds_conv_max_value' +str(max_value))
      int_bits = int(np.ceil(np.log2(max(abs(min_value),abs(max_value)))))
      dec_bits = 7-int_bits
      print('int_bits' +str(int_bits))
      print('dec_bits' +str(dec_bits))
    
    
    
    bn = tf.nn.relu(depthwise_conv)

saichand07, 2019-05-09:
> Sorry, my English is not good, I said how to calculate q2.9. From which value is obtained. weights don't need calculate.

I used min and max values of the weights, so that I calculated the Q format. As you can see the code above

saichand07, 2019-05-09:
act_max = [32,8,8,8,8,8,8,8,8,8,8,8], 
where, I got good results

saichand07, 2019-05-09:
@ccnankai 
thanks, I found the problem, weights are wrong
now its working

weishu20, 2019-06-24:
> @ccnankai
> thanks, I found the problem, weights were wrong
> now its working

hello，can you tell me how to calculate Q of input and output？I have no idea....

ccnankai, 2019-07-08:
@weishu20 
the first layer is input layer(mfcc)

closed 2019-07-08 by ccnankai