ARM-software/ML-KWS-for-MCU #97: Reasons for the Random keyword detection on board ?

saichand07, 2019-03-25:
Hello All,
I have deployed the DS-CNN model on S32K148 (external analog microphone connected), which is detecting some random output without any input given by user.
if we say a keyword loudly and multiple times then its working.
After tuning the frame_length, frame_shift,  num_frames mel low and mel upper frequency, it is detecting silence and unknown words with maximum accuracy and other keywords are haven't detecting properly?
 
it would be really helpful, if somebody know the answers to this problem

Is the problem with the
model ?

parameters (frame_length, frame_shift, num_frames,......etc) ?

Microphone (Currently I am using Analog, do I need to to switch to Digital)?

Thanks

pooyaww, 2019-03-26:
In real time audio signal classification there are many other issues which requires your sound and audio engineering knowledge.
Try tuning these tree parameters:
**_recording_win
averaging_window_len
detection_threshold
Also try tuning the microphone gain._**
at the end the system should work fine in close distance or in a open environment without too much noise.
in closed environment like a room you need to dereverb /denoise the audio signal by filtering, or using particular methods in training the network to improve the real-time performance.
Also, if you are using the sample data-set from google, you should consider it is not very well-made for real world applications.
All in all, there are several layers of tuning from Data, NN, SW to HW, taking into account that you have already achieved a good performance in an off-line test.

saichand07, 2019-03-26:
@pooyaww 
I take this answer, Thanks
Did you deploy the models on MCU, Is your models working well on MCU ?
what are the optimum values that did work well in your case?

ccnankai, 2019-03-27:
@saichand07 
I succeeded.Using the speech_momandv2 data set，adjust the act_max parameter layer by layer, even if the accuracy is at %80, you can see the result on mcu.

saichand07, 2019-03-27:
@ccnankai 
can you explain it shortly
how to calculate bias_shift, out_shift (left, right) manually?

ccnankai, 2019-03-27:
@saichand07 
Run quant_test.py, all parameters  are the same as those in train.py. At this time, set act_max=x, 0,0,0,0,0,0,0,0.... and select the value with the highest accuracy as the true value of x. Then run quant_test.py, act_max=x, y, 0, 0, 0, 0, 0, 0, and select the value with the highest accuracy as the y value. The greedy algorithm is carried out layer by layer.

saichand07, 2019-03-27:
@ccnankai Thanks
y value is also powers of 2 right ?
and we need to repeat to every layer ?

ccnankai, 2019-03-27:
@saichand07 
Powers of 2, traversing each layer, I get good results.

saichand07, 2019-03-27:
@ccnankai 
python models working really well, 
but the Deployed model on the board giving bad results,

saichand07, 2019-03-28:
@ccnankai 
#define CONV2_DS_BIAS_LSHIFT 2
#define CONV2_DS_OUT_RSHIFT 5
#define CONV2_PW_BIAS_LSHIFT 4
#define CONV2_PW_OUT_RSHIFT 8

do you know how to calculate the bias_shift, out_shift (left, right), where do we get those values?

saichand07, 2019-05-03:
Hai @navsuda @ccnankai @pooyaww 

I trained ds_cnn model with the same parameters they have mentioned in this tutorial, then after I tested and fused the checkpoints, therefore I ran the quant_test.py to generate the weights, [ I also tuned the act_max values layer by layer, i got good results ]

I also calculated the bias_shift and out_shift manually layer by layer, I also tuned the recording window, averaging window_len , detection threshold but the deployed model on board not detecting the correct words at all.
 
the weights and biases as follows,
int_bits-3
dec_bits10
DS-CNN_conv_1_weights_0 number of wts/bias: (10, 4, 1, 64) dec bits: 10 max: (0.087890625,0.08807392) min: (-0.111328125,-0.1116012)
int_bits1
dec_bits6
DS-CNN_conv_1_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.421875,1.4201941) min: (-1.03125,-1.033471)
int_bits3
dec_bits4
DS-CNN_conv_ds_1_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 4 max: (3.875,3.9047031) min: (-4.125,-4.1088867)
int_bits2
dec_bits5
DS-CNN_conv_ds_1_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (3.4375,3.4457488) min: (-1.59375,-1.5969778)
int_bits0
dec_bits7
DS-CNN_conv_ds_1_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.7578125,0.7612533) min: (-0.59375,-0.5959456)
int_bits2
dec_bits5
DS-CNN_conv_ds_1_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.75,2.7442703) min: (-1.75,-1.7491003)
int_bits2
dec_bits5
DS-CNN_conv_ds_2_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (2.6875,2.6751502) min: (-1.96875,-1.9610264)
int_bits1
dec_bits6
DS-CNN_conv_ds_2_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.46875,1.4665873) min: (-1.265625,-1.262942)
int_bits0
dec_bits7
DS-CNN_conv_ds_2_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.7890625,0.7914842) min: (-0.5703125,-0.569746)
int_bits1
dec_bits6
DS-CNN_conv_ds_2_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.96875,1.9751662) min: (-1.5625,-1.5675924)
int_bits2
dec_bits5
DS-CNN_conv_ds_3_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 5 max: (2.0,2.0118198) min: (-1.4375,-1.4240787)
int_bits2
dec_bits5
DS-CNN_conv_ds_3_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.0,2.0021482) min: (-2.40625,-2.3984358)
int_bits0
dec_bits7
DS-CNN_conv_ds_3_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.546875,0.5507306) min: (-0.5625,-0.5627646)
int_bits1
dec_bits6
DS-CNN_conv_ds_3_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.859375,1.8647689) min: (-1.5625,-1.5585374)
int_bits3
dec_bits4
DS-CNN_conv_ds_4_dw_conv_depthwise_weights_0 number of wts/bias: (3, 3, 64, 1) dec bits: 4 max: (4.125,4.1244545) min: (-3.1875,-3.1993992)
int_bits1
dec_bits6
DS-CNN_conv_ds_4_dw_conv_biases_0 number of wts/bias: (64,) dec bits: 6 max: (1.828125,1.8263378) min: (-1.921875,-1.9233539)
int_bits0
dec_bits7
DS-CNN_conv_ds_4_pw_conv_weights_0 number of wts/bias: (1, 1, 64, 64) dec bits: 7 max: (0.7734375,0.7710628) min: (-0.7265625,-0.72906214)
int_bits2
dec_bits5
DS-CNN_conv_ds_4_pw_conv_biases_0 number of wts/bias: (64,) dec bits: 5 max: (2.03125,2.023883) min: (-0.90625,-0.91236055)
int_bits3
dec_bits4
DS-CNN_fc1_weights_0 number of wts/bias: (64, 12) dec bits: 4 max: (2.25,2.261885) min: (-4.125,-4.1407604)
int_bits-1
dec_bits8
DS-CNN_fc1_biases_0 number of wts/bias: (12,) dec bits: 8 max: (0.375,0.37431768) min: (-0.1953125,-0.19587071)

Can you please give me your ideas or suggestions, to make this model run on board with correct keyword detection.

Thank you very much

pooyaww, 2019-05-04:
@saichand07 
Did you apply your calculated shift values on the MCU code?
`Deployment/Source/NN/DS_CNN/ds_cnn.h`

saichand07, 2019-05-04:
@pooyaww thanks for the reply
yes, I did apply shift values,

I calculated all the shifts like this
first layer input Q 5.2
weights Q-3.10
bias  Q1.6
output layer Q 3.4

out_shift = 2+10-4 = 8
bias_shift = 2+10-6 = 6

last layer bias shift I got negative shift and one of the middle layer bias_shift 0, 
Could you please tell me how to handle this ?

saichand07, 2019-05-10:
I figured out the problem, my weights are wrong, Now my model is working fine.
@pooyaww, @ccnankai thanks for your help

closed 2019-05-10 by saichand07