ARM-software/ML-KWS-for-MCU #102: about training samples.

ccnankai, 2019-04-19:
In the actual application scenario, do the training samples need to use far-field, near-field, and different orientation sounds? If the audio clips have different durations at the beginning, do you need to normalize these audio clips?

saichand07, 2019-04-23:
Hai @ccnankai 
Training samples can be a far-field, near-field and different orientation sounds, speech_commands_v0.02 dataset also has different orientation of sounds. 
In this tutorial they did mix the background noise and silence to the recordings, but afterwards they didn't do any audio clips normalization.
may be you can do with and without normalization

saichand07, 2019-04-23:
@ccnankai 
how did you deal with negative bias_shift and out_shift, I have calculated bias_shift and out_shift of every layer and I did put those numbers in c code but my deployed model showing really bad output. 
do you have any idea, what i need to tune or check ?

ccnankai, 2019-04-24:
Hi, @saichand07 
Do you use your own data? I use my own data and the results are not good.

saichand07, 2019-04-24:
@ccnankai 
How much data do you have? if it is very less, try to make your model small and train it.
I also used my own data but size of my dataset is very small, therefore results were poor.

saichand07, 2019-04-24:
> @ccnankai
> how did you deal with negative bias_shift and out_shift, I have calculated bias_shift and out_shift of every layer and I did put those numbers in c code but my deployed model showing really bad output.
> do you have any idea, what i need to tune or check ?

@ccnankai can you give me answer to this question?

ccnankai, 2019-04-24:
@saichand07 
I think your shift parameter is wrong. If it is correct, even if one or two bits are wrong, the result is acceptable.

saichand07, 2019-04-24:
I calculated shift parameters as they explained in the tutorial,
act_max = [32 8 8 8 8 8 8 8 8 8 8 8 ]

first input layer Q5.2
weights Q-2.9    (do we need to take -9 dec bits or 9)
Bias Q1.6
output Q3.4 (do we need to take last layer or next layers)

bias_shift_left = 3
out_shift right = 7

I have calculated same way for all layers