ARM-software/ML-KWS-for-MCU #111: Saving checkpoints in test.py

marjanemd, 2019-07-08:
I am having some issues with saving the checkpoints in:
 python test.py --model_architecture dnn --model_size_info 128 128 128 --checkpoint 
<checkpoint path>

Where and how should I save the checkpoint here?

eera-l, 2019-07-09:
Hi @marjanemd, 
you can manually set where you want to save the checkpoints when you train the models using `--summaries_dir` and `--train_dir`. To see some examples, check the commands on the train_commands file:
[https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/train_commands.txt](url)

If you use the commands in that file, your checkpoints will be saved in the /work folder. 
If you want to test the trained models using their checkpoints, use for example:

`python test.py --model_architecture dnn --model_size_info 128 128 128 --checkpoint work/DNN/DNN1/training/best/dnn_9282.ckpt-4000`

marjanemd, 2019-07-09:
Thanks @eera-l .
I also do not know how can I have access to see the folders inside the work folder. They are not in the ML-KWS-for-MCU folder that was cloned. Are they saved somewhere local?

eera-l, 2019-07-10:
The work folder and subfolders are created automatically when you first train your models. 
Once you have finished training the models, you will have a few checkpoints in the work subfolders.

You get that error because you probably don't have that checkpoint on your computer (it was just an example to show what the command should look like). You will have to change the path to your checkpoints once they are created. :)

marjanemd, 2019-07-10:
Thank you very much @eera-l . Right now it does not give me error for the checkpoints, but I have 2 questions:

1. There are several  (18) checkpoints with different suffixes created, (.data, .index, .meta), how should I know which to use?  I am using the last one with .data right now. But I believe it is not right since I am getting an error related to mismatch between the current graph and the other one,
 The following is the error, really appreciated if you could help me with this:

  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/training/saver.py", line 1322, in restore
    err, "a mismatch between the current graph and the graph")
tensorflow.python.framework.errors_impl.InvalidArgumentError: Restoring from checkpoint failed. This is most likely due to a mismatch between the current graph and the graph from the checkpoint. Please ensure that you have not altered the graph expected based on the checkpoint. Original error:

Assign requires shapes of both tensors to match. lhs shape= [128] rhs shape= [144]
	 [[node save/Assign_6 (defined at /content/ML-KWS-for-MCU/models.py:159) ]]

marjanemd, 2019-07-10:
Thanks for your help @eera-l

closed 2019-07-10 by marjanemd