ARM-software/ML-KWS-for-MCU #118: 【Help】If i train model with training data ,which is 1.5s not 1s. How many paras should be changed?

xjj210130, 2019-11-21:
Hi  all:
   when i use the cmd:
  **python train.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --learning_rate 0.0005,0.0001,0.00002 --how_many_training_steps 10000,10000,10000 --summaries_dir work/DS_CNN/DS_CNN1/retrain_logs --train_dir work/DS_CNN/DS_CNN1/training** 

The  training data is speech_commands_v0.02.  All data time is 1seconds.  And use 40 MFCC features extracted from a speech frame of length 40ms with a stride of 20ms, which gives 1960 (4940) features for 1 second of audio.

**However, my own training sample is 1.5 seconds long. When I use the above command to train, which parameters need to be modified?**
Thanks very much

xjj210130, 2019-11-21:
Only need to modify the parameters clip_duration_ms=1500?

dimanshu, 2020-06-01:
@xjj210130  did u solve it ?