ARM-software/ML-KWS-for-MCU #120: ValueError : The passed save_path is not a valid checkpoint

Shenfanlin, 2019-11-28:
my checkpoint path is absolute path but it is valueerror when i ran the freeze.py in the ML-KWS-for-MCU.How can i solve this problem?

rookie-may, 2020-01-15:
I also have the same problem, have you solved it?

rookie-may, 2020-01-15:
> my checkpoint path is absolute path but it is valueerror when i ran the freeze.py in the ML-KWS-for-MCU.How can i solve this problem?

I also have the same problem, have you solved it?

dimanshu, 2020-05-18:
@rookie-may  did u solve it?

maxbalrog, 2020-09-18:
@dimanshu I've had the same issue, here is what solved it for me. Hope this might be helpful.

In my checkpoint folder (.../ckpt/), I've had 3 files: 
```
<model_name>.ckpt-30000.index
<model_name>.ckpt-30000.meta
<model_name>.ckpt-30000.data-00000-of-00001
```

The script freeze.py worked just fine for me when I ran it with flag `--checkpoint <path_to_your_ckeckpoint_folder>/ckpt/<model_name>.ckpt-30000`