ARM-software/ML-KWS-for-MCU #122: quant_test with different act_max parameters but produces same content weights.h ?

jessonwtp, 2020-01-10:
Hi All,

When I use quant_test.py to tune act_max parameters (one by one with value 8, 16, 32), I can see the training accuracy, validation accuracy and test accuracy are changing according to different act_max parameters.
But, why I got the same weights.h ? Is it reasonable ?
I can get good accuracy by test.py and label_wav.py.
But when I use this quanted weights file to my embedded device, the test result is great different. 
that's why I think maybe caused by quant function.

It's appreciate is anyone knows the reason and share your experience. 
Thanks.

ShintaZachariah, 2020-01-16:
Hi All,
what is act_max parameter?

jessonwtp, 2020-01-21:
Hi Shintzchariah,
I use 32 0 0 0 0 0 0 0 0 0 0 0 & 16 0 0 0 0 0 0 0 0 0 0 0 & 32 8 8 32 16 8 8 8 8 8 32 16.

jessonwtp, 2020-01-21:
I'm trying the quant bias shift and out shift which set in the embeded Device side.

closed 2020-02-04 by jessonwtp