ARM-software/ML-KWS-for-MCU #140: update to TensorFlow 2

upcoders, 2020-07-18:
Is there an update to work with tensorflow version 2, especially for train.py?

Thanks

Burton2000, 2020-11-09:
See https://github.com/ARM-software/ML-examples/tree/master/tflu-kws-cortex-m for tf2 version of training and quantizing.

Aishaj, 2021-07-22:
@Burton2000 Thanks, but how can I use the pre-trined models in the TensorFlow2 code?

Burton2000, 2021-07-26:
@Aishaj You can use them with the test_tflite.py file to evaluate the model accuracy on the test set, check the README in the Training folder. You need to make sure the same arguments from training are used with test_tflite (see train_commands.txt)

Aishaj, 2021-07-27:
@Burton2000 Thanks, we would also appreciate adding label_wav.py file for TensorFlow 2 version. THank you

Burton2000, 2021-07-27:
Yep if that is something that will be of use we will look to add that as well when we have some time.