ARM-software/ML-KWS-for-MCU #145: mbed error

aktoey, 2021-08-11:
$ mbed new kws_simple_test --mbedlib
[mbed] Working path "E:\Zbit\AI\KWS\Git\ML-KWS-for-MCU\Deployment" (program)
[mbed] Creating new program "kws_simple_test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Unpacking library build "65be27845400" in "E:\Zbit\AI\KWS\Git\ML-KWS-for-MCU\Deployment\kws_simple_test\mbed"
[mbed] Unpacking library build "65be27845400" in "E:\Zbit\AI\KWS\Git\ML-KWS-for-MCU\Deployment\kws_simple_test\mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/65be27845400"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

aktoey, 2021-08-11:
What is the reason for this？

EmbeddedSysDesign, 2023-01-11:
Hello,

I am trying to compile an old project which is available on github https://github.com/ARM-software/ML-KWS-for-MCU. The original project was using mbedOS 2 that I know has deprecated (https://os.mbed.com/handbook/Help). But this is not my problem as you can find in the attached compilation report, program is failing at 98.5% at CMSIS-NN repositories/files. I hope ARM support will look into this issue and guide me to solve this problem. I am very close to this if you can give me the fix or let me know what should I do to build the [[Hello Edge: Keyword spotting on Microcontrollers]](https://arxiv.org/pdf/1711.07128.pdf).

Thank you,
FS 

Project details: Hello Edge: Keyword spotting on [Microcontrollers](https://github.com/ARM-software/ML-KWS-for-MCU)
STM32F746 Disco board
mbed-cli v1.8.3
Python 2.7.x
mbedos 2 library files

********************************** Error****************************
[Error] arm_acle.h@48,49: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
[Error] arm_acle.h@55,50: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
[Error] arm_acle.h@62,49: invalid conversion from 'void*' to 'int*' [-fpermissive]
[Error] arm_acle.h@69,50: invalid conversion from 'void*' to 'int*' [-fpermissive]
[Error] arm_acle.h@100,50: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
[Error] arm_acle.h@107,51: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
[Error] arm_acle.h@114,50: invalid conversion from 'void*' to 'int*' [-fpermissive]
[Error] arm_acle.h@121,51: invalid conversion from 'void*' to 'int*' [-fpermissive]
[Error] nn.h@28,25: 'q7_t' has not been declared
[Error] nn.h@28,40: 'q7_t' has not been declared
[Error] kws.cpp@72,33: no matching function for call to 'NN::run_nn(q7_t*&, q7_t*&)'
[Error] kws.cpp@74,47: 'arm_softmax_q7' was not declared in this scope
[ERROR] In file included from ../CMSIS-NN/Include/arm_nn_math_types.h:94:0,
                 from ../CMSIS-NN/Include/arm_nnfunctions.h:114,
                 from ../Source/NN/nn.h:22,
                 from ../Source\KWS\kws.h:23,
                 from ../Source\KWS\kws.cpp:24:
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_ldc(unsigned int, unsigned int, const void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:48:49: error: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
   return __builtin_arm_ldc (__coproc, __CRd, __p);
                                                 ^
<built-in>: note:   initializing argument 3 of 'void __builtin_arm_ldc(unsigned int, unsigned int, const int*)'
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_ldcl(unsigned int, unsigned int, const void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:55:50: error: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
   return __builtin_arm_ldcl (__coproc, __CRd, __p);
                                                  ^
<built-in>: note:   initializing argument 3 of 'void __builtin_arm_ldcl(unsigned int, unsigned int, const int*)'
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_stc(unsigned int, unsigned int, void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:62:49: error: invalid conversion from 'void*' to 'int*' [-fpermissive]
   return __builtin_arm_stc (__coproc, __CRd, __p);
                                                 ^
<built-in>: note:   initializing argument 3 of 'void __builtin_arm_stc(unsigned int, unsigned int, int*)'
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_stcl(unsigned int, unsigned int, void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:69:50: error: invalid conversion from 'void*' to 'int*' [-fpermissive]
   return __builtin_arm_stcl (__coproc, __CRd, __p);
                                                  ^
<built-in>: note:   initializing argument 3 of 'void __builtin_arm_stcl(unsigned int, unsigned int, int*)'
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_ldc2(unsigned int, unsigned int, const void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:100:50: error: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
   return __builtin_arm_ldc2 (__coproc, __CRd, __p);
                                                  ^
<built-in>: note:   initializing argument 3 of 'void __builtin_arm_ldc2(unsigned int, unsigned int, const int*)'
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h: In function 'void __arm_ldc2l(unsigned int, unsigned int, const void*)':
c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\lib\gcc\arm-none-eabi\6.3.1\include\arm_acle.h:107:51: error: invalid conversion from 'const void*' to 'const int*' [-fpermissive]
[... cut ...]
       Path: "C:\prj_KWS\Deployment\ML-KWS-for-MCU\Deployment\kws_realtime_test"
       Command: "C:\Python27\python.exe -u C:\prj_KWS\Deployment\ML-KWS-for-MCU\Deployment\kws_realtime_test\tools\make.py -t GCC_ARM -m DISCO_F746NG --profile ../release_O3.json --source . --source ../Source --source ../Examples/realtime_test --source ../CMSIS-NN/Include --source ../CMSIS-NN/Source --source ../CMSIS-DSP/Include --source ../CMSIS-DSP/PrivateInclude --source ../CMSIS-DSP/Source --source ../CMSIS_5/CMSIS/Core/Include --build .\BUILD\DISCO_F746NG\GCC_ARM-RELEASE_O3 -j 8"